mavishak/cnss-embedded #9: Using the HC-SR501 PIR Sensor

mavishak, 2020-12-21:
**_21.12.2020:_** _Sensor works using D6 interrupt and LD2 as a response._

**TBD:**
- [ ]  Change D6 to D5 (PB4).
- [ ]  change the response to USART.
- [ ]  Play with the sensor configurations.

**These are the documents and links we have used in the learning process.**
[HC-SR501 PIR Sensor](https://components101.com/hc-sr501-pir-sensor)
[How HC-SR501 PIR Sensor Works & Interface It With Arduino](https://lastminuteengineers.com/pir-sensor-arduino-tutorial/) _- very informative._

[HC SR501 PIR Sensor Datasheet.pdf](https://github.com/mavishak/cnss-embedded/files/5725492/HC.SR501.PIR.Sensor.Datasheet.pdf)
[PIR Motion Sensor.pdf](https://github.com/mavishak/cnss-embedded/files/5725493/PIR.Motion.Sensor.pdf)

mavishak, 2020-12-27:
### Problem with USART:

  **23.12.2020:** 
We debugged usart_init() and write() it seems that they work as they are supposed to
   			But we haven't managed to view the transmitted letter on the screen.
   			Need to dig deeper - and solve the problem.
 
   **27.12.2020:** 
When debugging write() USART2->DR and TXE flag in USART2->SR are not changing. 
                        This might be normal.
                        There might be a problem with ST-LINK/v.2
   			Note: Changed to branch main_v.2 because of problems in the main branch

mavishak, 2020-12-27:
**27.12.2020:** 
_We managed to connect TeraTerm and USART2 by enabling PA2._
**Problem:** TeraTerm is not displaying character correctly.

mavishak, 2020-12-27:
_Tera Term still printing garbege._ 
Got Interrupt Tx to work. 
TC interrupt should come next.

mavishak, 2020-12-30:
## USART Working

**30.12.2020:** 
_**_Good News:_** Got TeraTerm To work!
**_Bad News:_** I have no idea why it's working... :/_
Basically, the problem is in the setting of the BAUD RATE.
We need to get a better understanding of its calculation and of the clock tree.

**_How is it working:_**
TeraTerm speed is set to 2400 and USART_BRR is set to 0xD05.
(I don't know if it matters but - GPIOA is configured to a speed of 2 MHz.)
Note: Interrupt is now disabled - Instructions from above.

**_I am adding 3 images displaying what worked:_** 
_(click to make larger)_

[image]
[image]
[image]

mavishak, 2020-12-30:
## _**One other thing to take note of:**_
I didn't notice before but it seems that in the working version of USART RCC_AFIO is disabled _**(or is it really?)**_.

The reason behind it is as we were debugging we noticed that the configuration of RCC_AFIO does not affect the garbage displayed on the screen, so we commented it out _**(and forgot about it).**_ 

So I guess another question would be: **When does RCC_AFIO need to be enabled?**

mavishak, 2021-01-03:
**03.01.2021:** _(hc-sr501pir_sensor.c)_ 
USART2 is now configured to respond to the sensor interrupt.
 **_TBD:_** Implement triggering of USART with event_queue.
 **_Question:_** Should the handler init() the uart and then trigger write() or should UART be initialized in advance.
 My gut tells me that the handler should do both...

**_03.01.2021:_** _(usart.c)_
A message is now sent to the screen rather than one letter.
The problems listed above are still valid :/

## Images for Documentation:
 **_HC-SR501 PIR Sensor_**
_(click to make larger)_

[image]
[image]
[image]
[image]

**_The sensor is now connected to USART. The next step is to connect the STM32 and ESP8266 Wi-Fi module._**

mavishak, 2021-01-08:
**08.01.2021:**
 _**SOLVED! :)  USART IS NOW OFFICIALLY WORKING with a baud rate of 9600 bps.** 
   The USART2 clock is PCLK1 which is by default only 8MHz NOT 36MHz.
 			   (see RM p.798 for BRR calculation and RM p.93 Fig.8 for clock tree).
 			   So, basically, our problem was that we thought that PCLK1 was set to 36MHz by default.
 			   For later practice follow the clock tree and default initialization of clock registers._

NaomiCreate, 2021-01-17:
[image]