mavishak/cnss-embedded #15: Sending HTTP request from STM32 to Firebase via ESP8266

mavishak, 2021-02-28:
Some Pointers:

- It seems that Cloud Firestore can't be used for embedded devices. _[Integration with Firestore?](https://github.com/mobizt/Firebase-ESP8266/issues/126)_

mavishak, 2021-03-04:
## We managed to send a post msg to the Real-time database in Firebase

**04.03.2021:** 

**We sent a POST HTTPS request using port 443 and SSL**
1. We needed to write: _AT+CIPSSLSIZE=4096_ [ESPRESSIF: Board index English Forum Explore General Discussion](https://www.esp32.com/viewtopic.php?t=12746)
2. We needed to write: _sprintf((char*)command, "AT+CIPSTART=\"SSL\",\"%s\",%ld\r\n",(char*)firebase_host, https_port)_ [StackExchange: Making HTTPS requests using ESP8266 AT commands](https://arduino.stackexchange.com/questions/80019/making-https-requests-using-esp8266-at-commands)
3. The POST message was:
> POST /rest/test/posts.json? HTTP/1.1
> Host: ****
> Content-Type: application/json
> Content-Length: 49
> 
> {"author": "Nemo Resh", "title": "Fish are blue"}

_Note: we got the format of the HTTP message using [Postman](https://www.postman.com/) and Firebase documentation [Firebase: Saving Data](https://firebase.google.com/docs/database/rest/save-data)_

mavishak, 2021-03-04:
## Moving Forwards

The HTTP request to firebase should be as follows:

> PUT /devices/device_1/history/04-03-2021/21:48.json?auth=[DATABASE_SECRET] HTTP/1.1

> Host: [DATABASE_HOST]
> Content-Type: application/json
> Content-Length: [BODY LENGTH]
> 
> {"image_path": "image", "notes": "alarm went off"}

_see: [How do I authenticate my Firebase with just a string key](https://stackoverflow.com/questions/50083815/how-do-i-authenticate-my-firebase-with-just-a-string-key)_

## A Few Issues

- user must be able to change Wi-Fi SSID and PSW without passing through the web.
There are two ways to achieve this: 
1. usart2 - this is probably possible
2. Key matrix - this is probably preferable

- device authentication will be done using the database secret and device id.

mavishak, 2021-03-05:
We need to find a way to get the actual date and time when an alert is created.
 We don't expect <time.h> to work because this is an embedded system, but we checked it anyways:
see: [c function localtime](https://www.tutorialspoint.com/c_standard_library/c_function_localtime.htm)

```
void timestamp(void){

	   time_t rawtime;
	   struct tm *info;
	   time( &rawtime );
	   info = localtime( &rawtime );
	   memset((char*)command, '\0', COMMAND_SIZE*sizeof(uint8_t));
	   sprintf((char*)command, "Current local time and date: %s", asctime(info));
	   write_usart2((uint8_t*)command);

}
```
The output time and date was not correct:

>  _Current local time and date: Wed Dec 31 23:59:59 1969_

We tried playing with firebase server value - timestamp:

> _{"image_path": "image", "notes": "alarm went off!", "time": {".sv": "timestamp"}}_

We did not find a way to insert the timestamp in the path and it seems the timestamp does not indicate the actual time and date as it is **the time since UNIX epoch, in milliseconds**.
see: [Firebase Timestamp](https://firebase.google.com/docs/reference/android/com/google/firebase/Timestamp)
and: [Firebase server values Reat API](https://firebase.google.com/docs/reference/rest/database#section-server-values)

mavishak, 2021-03-06:
### 06.03.2021
After consulting our instructor we have decided to use the  Firebase timestamp and convert it to the date and time using:
UNIX timestamp converters.

> var s = new Date(1504095567183).toLocaleDateString("en-US")
// expected output "8/30/2017"  
console.log(s);

see:
[convert UNIX timestamp to date and time](https://www.codegrepper.com/code-examples/javascript/convert+unix+timestamp+to+datetime+formula+in+angular)
[convert timestamp to date and time in angular](https://stackoverflow.com/questions/21848407/convert-time-stamp-to-date-inside-angular-expression-like-new-datetimestamp)

NaomiCreate, 2021-03-07:
### 07.03.2021
We organized the code. Now the alert is recorded under the device-ID in the Real Time Database.
!TBD:

- [ ] AT+RST is very important'but it doesn't work properly -> need to find a way to make it work
- [ ] Need to create a systematized way to deal with: PASS, FAIL and INFINITE LOOPS.
- [ ] Some of the PASS\FAIL responses are irrelevant, and could cause problems-> go over them, carefully.
- [ ] Should we use on the HTTP request: `print=silent`?
- [ ] Need to minimize constant arrays holding Strings. (e.g. content array)