mavishak/cnss-embedded #16: Enabling TIM2

mavishak, 2021-03-17:
## 17/03/2021
We started implementing timer2. As of now, it seems that TIM2->CNT changes its value as expected.
We couldn't be sure because it wouldn't print in UART. 
At first, ARR wouldn't change in debugging mode but then we set the (=) value rather than the |= operator.
We decided to move forward and implement the timer interrupts.

[Stm32 Intro To timers - YouTube](https://www.youtube.com/watch?v=BtAi6-7Lnlw)
[Stm32 Delay Using Timers or Systick - YouTube](https://www.youtube.com/watch?v=2YSYWR309Y4)

NaomiCreate, 2021-03-20:
### 20/03/2021
We implemented timer 2 ,timer 3 and timer 4. 
And we created for all of them delay functions and timeout functions.
Timers 2 and 3 fire an interrupt once a milli second.
Timer 4 fire an interrupt once a second.

mavishak, 2021-03-23:
##24/03/2021

recordAlert() is now working with timers. We changed the HTTP version to 1.0 to get a quicker CLOSING. When sending an image we will probably need version 1.1