mavishak/cnss-embedded #18: Testing

mavishak, 2021-03-24:
### 24/03/2021
Today we let the device run for a few hours and recorded the last hour's logs - the file is attached below.
Pass is identified with :)
Fail is identified with :(

The AT commands that where problematic:
_AT+CIPSTART="SSL","firebase-url.com",443
AT+CWMODE=1_ 

They might need more time or we might need to disconnect with a CIPCLOSE after CIPSTART even if it fails.
[1 hout run output - 17-03-2021.log](https://github.com/mavishak/cnss-embedded/files/6198458/1.hout.run.output.-.17-03-2021.log)

NaomiCreate, 2021-03-26:
### 26/03/2021

**We recorded an hour log of the following main:**
```
#include "stm32f103xb.h"
#include "hc-sr501pir_sensor.h"
#include "usart.h"
#include "esp8266_Firebase.h"
#include "esp8266_WiFi.h"
#include "timers.h"
#include "common.h"

int main(void)
{
init_usart2();
init_timer4();
init_usart1();
write_usart2((uint8_t*)("\r\n_______________\r\n"));
while(1)
{
if(recordAlert()){
write_usart2((uint8_t*)(":)\r\n"));
}
else{
write_usart2((uint8_t*)(":(\r\n"));
}
write_usart2((uint8_t*)("\r\n.....................\r\n"));
write_usart2((uint8_t*)("\r\n\r\n"));
}
}

```

**The delay times and timeout in the test were:**
_ssl_tries = 3
_cipstart_tries = 3
_ssl_timeout = 6
_cipstart_timeout = 20
We set them in the following call to function:
`connectFirebaseHost(3,3,6,20)`

The alerts are sent to firebase under the id: **Timon-device-ID**

mavishak, 2021-03-29:
### 29/03/2021
After analyzing both log files, we found that setting the timeout value, when connecting to the Firebase host, as described above seems to do the trick. In one hour we sent 549 alerts in firebase, not one of them failed.

_We need to test a few more times to make sure that it was not pure luck..._