mavishak/cnss-embedded #30: GET/POST with pausing time

mavishak, 2021-05-16:
### 16/05/2021
Sometimes communication gets stuck in the context of GET and / or POST requests.

Explanation: 'GET', this is  request asking the database whether the system is turned ON or OFF (i.e. does or does not receive alerts, respectively). This request is triggred periodicaly  by the system to check whether the configurations required by the user has changed. There is a paussing time count performed by TIMER2.

'POST' this when the system detects traffic, and spontaneously sends a message to the database. After sending such a message, there is a period of time in which we are not interested in further traffic manifestations. This time period is also counted by timer, this time TIMER3.

TIMER4 - Its role is to schedule in communication with the ESP. If it takes too long for a response from the ESP, the timer expires and a proper response must be decided.

 
Our suspicion falls on the multiplicity and length of interrpts and events.

mavishak, 2021-05-16:
ESP8266 as a server:

From what we understoof inorder to configure the ESP as server we needed to configure it as an access point (AP).

> AT+CWMODE=2 // configure as AP
> AT+CWSAP=<AP_ssid>,<AP_pwd>,<chl>,<ecn>[,
> <max conn>][,<ssid hidden>] // from what we saw we could only set the password when using the given ssid (AT+CWSAP?)

After processing thes two AT commands we could access the AP from out computer/smartphone

> AT+CWLIF // prints out all the current IP's that are connected to the ESP access point.

We didn't quite understand if this could actually help us, but we will look further in to the subject.
_A helpful link - [ESP8266 AP & STA MODE TOGETHER](https://siytek.com/esp8266-ap-and-station-mode/)_

NaomiCreate, 2021-05-31:
Our instructor recommended that we reduce the number of prints to Tera Term ,so that we can dubug the software more efficiently. Therefore we changed the implementation of uart2 so that printing to Tera Term will occur after each line only.

We continued with a debugging session that included tests for the timers, and the motion sensor.
Attached is the documentation file of the debugging session:

[Debug session 23.5.21-24.5.pdf](https://github.com/mavishak/cnss-embedded/files/6570164/Debug.session.23.5.21-24.5.pdf)

**The bug we found was:**
We cleaned the sensor interrupt bit only when timer 3 came to a timeout, instead of clearing that bit at the beginning of the interrupt.
**With bug:**
```
void EXTI4_IRQHandler(void) 
{ 	
       if(timeout_done_timer3()) 	
       { 		
                EXTI->PR |= 0x00000010;		
		add_event(alert_Handler); 		
                set_timeout_timer3(10);  	
        }
}
```

**The fix:**
```
void EXTI4_IRQHandler(void) 
{ 	
       EXTI->PR |= 0x00000010;	
       if(timeout_done_timer3()) 	
       { 			
		add_event(alert_Handler); 		
                set_timeout_timer3(10);  	
        }
}
```