mavishak/cnss-embedded #36: SSL connection not established

mavishak, 2021-07-05:
**06-07-2021**
Up to a few days ago our ESP connection with firebase was wroking great. Then we didn't use the device for a few day and WAM the device is not able to connect with firebase. 

We are getting:

> AT+CIPSSLSIZE=4096
> 
> OK
> AT+CIPSTART="SSL","project-host",443
> 
> ERROR
> CLOSED

Once we got (this might not be the exact words):

> CONNECTION ALREADY ESTABLISHED 
> ERROR
> CLOSED

mavishak, 2021-07-06:
**06-07-2021**
_Here is a documentation of AT+CIPSTART
[https://docs.espressif.com/projects/esp-at/en/latest/AT_Command_Set/TCP-IP_AT_Commands.html#cmd-start](https://docs.espressif.com/projects/esp-at/en/latest/AT_Command_Set/TCP-IP_AT_Commands.html#cmd-start)_

**Try List:**

- [ ] RESET device - an actual reset not just AT+RST
- [ ] Monitor device SSL handshake with firebase using comunication monitorring tools
- [ ] Enlarge SSL buffer
- [ ] Change ESP baude rate to 9600 - needs to be done both ways, probably not a great option...
- [ ] Contact firebase

mavishak, 2021-07-06:
**real main backup**
```
int main(void)
{
	//set_sys_clock_to_32MHz();
	//init_MCO();

	USART1_init(); // for ESP8266
	USART2_init(); // for debugging

	TIMER2_init(); // for monitoring switch state.
	TIMER3_init(); // for sensor delay
	TIMER4_init(); // for ESP8266 timeout

	QUEUE_init();

	CONFIGURATIONS_set_device_id();

	USART2_enable_Rx(); // for ESC

	USART2_write((uint8_t*)("\033[32m\r\nPress ESC to configure\033[0m"));
	TIMER4_set_timeout(60);
	while(QUEUE_isEmpty() && !TIMER4_timeout_done());

	SENSOR_init(); // sensor interrupts are not inabled

	LED_init();

	//init_i2c1();
	USART2_write((uint8_t*)("\r\n_______________\r\n"));//For test

	while(1)
	{

		QUEUE_do_event();
		SYSTEM_CONTROL_monitor_switch_state(120); // every 2 minutes

	}
}
```

mavishak, 2021-07-06:
**Checking valid connection with wether API**
_This workd!_

Functionality in use:
```
int main(void){

	USART1_init(); // for ESP8266
	USART2_init(); // for debugging

	TIMER2_init(); // for monitoring switch state.
	TIMER3_init(); // for sensor delay
	TIMER4_init(); // for ESP8266 timeout

	//restore(1,60);// ESP8266 Restore factory flash setup
	localIP(2,6); // ESP8266 FOR MINITORRING PRINT LOCAL IP - DEBUGGING
	while(1){

		USART2_write((uint8_t*)("\r\n_______________\r\n"));//For test

		TestWifiConnection(); // TESTING WIFI CONNECTION

	}
}
```

```
/*this function is ment for testing WiFi conditions with TCP connection using whether API*/
BOOL TestWifiConnection(void){

	// before useing this function init_usart1(); and  init_usart2(); must be executed

	//Set client mode
	if(!setClientMode(2,6)){
		return FALSE;
	}
	USART2_write((uint8_t*)"1\r\n");

	//Join access point
	if(!joinAccessPoint_T(2,10)){
		return FALSE;
	}
	USART2_write((uint8_t*)"2\r\n");

	/*Default: AT+CIPMUX=0 (according to: AT instruction set- 5.2.15)*/

	//Connect HOST IP
	if(!connectWeatherAPI(2, 30)){
		return FALSE;
	}
	USART2_write((uint8_t*)"3\r\n");

	createTestMsg();
	USART2_write((uint8_t*)"4\r\n");

	//Send number of data bytes
	if(!sendRequest(1,1,30,40)){
		connection_closed = closeConnection(2,6);
		return FALSE;
	}
	USART2_write((uint8_t*)"5\r\n");

	USART1_write((uint8_t*)http);

	//Read response
	if(!readResponse(60)){//timeout set t0 3 minutes
		connection_closed = closeConnection(2,6);//added 30.4.21
		return FALSE;
	}

	USART2_write((uint8_t*)"6\r\n");

	//Close cunnection with firebase - this might be useless as firebase already closes connection with "CLOSED" response
	closeConnection(3,3);
	USART2_write((uint8_t*)"7\r\n");

	return TRUE;

}
```

Special functions:
- joinAccessPoint_T is basically the same as joinAccessPoint just with SSID__T and PWD_T
- connectWeatherAPI(2, 30)
```
BOOL connectWeatherAPI(uint32_t tries, uint32_t timeout){

	//Connect to API
	memset((char*)command, '\0', COMMAND_SIZE*sizeof(uint8_t));
	sprintf((char*)command, "AT+CIPSTART=\"TCP\",\"%s\",80\r\n", (char*)weather_api);

	found = STANDBY;
	USART1_write((uint8_t*)command);
	while(tries > 0){
		TIMER4_set_timeout(timeout);
		while(found == STANDBY && !TIMER4_timeout_done()){
			if(USART1_NEW_LINE_FOUND_get()){
				found = USART1_search_buffer_Rx((uint8_t *)AT_OK, (uint8_t *)AT_ERROR);
				if(found == STANDBY){
					found = USART1_search_buffer_Rx((uint8_t *)AT_ALREADY_CONNECTED, (uint8_t *)AT_ERROR);// CRITICAL!
				}
				USART1_NEW_LINE_READ_set();
			}
		}
		if(found == PASS){
			return TRUE;
		}
		else{ // FAIL OR TIMEOUT
			tries--;
			found = STANDBY; // reset found
			USART1_write((uint8_t*)command);
		}
	}
	return FALSE;

}
```
- createTestMsg
```
void createTestMsg(void){

	//Set HTTP request
	memset((char*)http, '\0', HTTP_SIZE*sizeof(uint8_t));
	sprintf((char*)http,("GET /data/2.5/weather?q=London,uk&appid=%s HTTP/1.0\r\nHost: %s\r\n\r\n\r\n"),weather_api_key,weather_api); // HTTP/1.0- Allow only one request
	http_len = strlen((char*)http)-strlen("\r\n"); // the last \r\n is for the AT command, and not included in the request's length

}
```

_**The rest is the same**_

mavishak, 2021-07-10:
**11-07-2021**
_**We managed to do an SSL handshake with our college domain. Firebase still not working. We suspect that the SSL buffer is no longer large enough or that the certification of the ESP8266 has expired**_

mavishak, 2021-07-11:
**12-07-2021**
It turns out we have two types of ESP8266 boards one is ESP-01 (which we used up until now) the other is  ESP-01S wich seems to work even worse :)
_see: [https://www.esp8266.com/viewtopic.php?t=11657
](https://www.esp8266.com/viewtopic.php?t=11657)_

- We connected the device directly to the PC this did not help.
We concluded the problem is not in our code:
1. It could be a Firebase new SSL configuration that we are not aware of
2. It could be an expired SSL certificate
3. It could be the hardware
4. It could be one or two or all three :)

**Some other things we tried:**
- Hardcoding the IP destination address
- Opened a new Firebase project (free)
- Tried creating a proxy using Postman - didn't seem to work, we think
-  it does not capture the WiFi module traffick

NaomiCreate, 2021-07-12:
We created a channel in [ThingSpeak](https://thingspeak.com/), and we managed to send POST request to it. (with TCP connection). (There is a graphic display in ThingSpeak that the request received)

mavishak, 2021-07-18:
**During the last week we have been reaching out to different sources:**

Here are the questions we posted

-  Stack overflow: _[https://stackoverflow.com/questions/68274011/ssl-handshake-with-firebase-realtime-db-stopped-working-sequentially-to-a-blaze](https://stackoverflow.com/questions/68274011/ssl-handshake-with-firebase-realtime-db-stopped-working-sequentially-to-a-blaze)_

mavishak, 2021-07-18:
For the last 5 months, we have successfully managed to create an SSL connection to Firebase via ESP8266 (ESP-01).
A week and a half ago the connection stopped working aside for a few individual times, a day after we couldn't connect at all.
-> AT+CIPSTART responded ever since with ERROR CLOSED.

We tried switching between different WiFi networks and got the same results.

[b]Details:[/b]
1. We are managing to create TCP connections with HTTP websites.
2. The WiFi module is connected to an STM32, the communication is established using AT commands via UART (baud rate 15200).
3. We tried issuing AT commands using a direct UART connection between the STM32  and the PC (independent of our code). The commands were given through a Terminal window in our PC - AT+CIPSTART still failed (same error).
4. We have 4 different WiFi modules - ESP8266 (ESP-01). They all reacted the same.
5. In the past we had a similar issue, back then we set the SSL size to 4096 (the maximum size and it helped).
6. We managed to connect to other HTTPS websites using an SSL connection
-> when connecting to these sites via AT+CIPSTART we got a few times the response CONNECTED OK, for the rest we got CLOSED OK.

[b]AT commands details[/b]
AT+RST (reset the module)
AT+CWMODE=1 (set the Wi-Fi mode to station mode)
AT+CWJAP="SSID","PWD" (connect to a Wi-Fi network)
AT+CIFSR (get the Wi-Fi module IP)
AT+CIPSSLSIZE=4096 (change the SSL buffer size)
AT+CIPSTART="SSL","project-name-rtdb.firebaseio.com",443 (establish a SSL connection)
AT+CIPSEND=300 (send data)

[b]Devices[/b]
Wi-Fi module: [url]https://www.makerfocus.com/products/4pcs-esp8266-esp-01-serial-wireless-wifi-transceiver-module-compatible-with-arduino[/url]
STM32 microcontroller: [url]https://www.st.com/en/evaluation-tools/nucleo-f103rb.html[/url]

[b]Reference[/b]
[url]https://github.com/espressif/esp-at/blob/v2.0.0.0_esp8266/docs/ESP_AT_Commands_Set.md[/url]
[url]https://www.espressif.com/sites/default/files/documentation/4a-esp8266_at_instruction_set_en.pdf[/url]

[b]Screenshots[/b]
AT commands
[image]

ESP8266 Images
[image]
[image]
[image]

If anyone can help, we would appreciate it very much.

Thank you in advance!

NaomiCreate, 2021-07-19:
**Documentation of successful SSL connection:**
[1 hour run output 29-06-2021.log](https://github.com/mavishak/cnss-embedded/files/6838462/1.hour.run.output.29-06-2021.log)

NaomiCreate, 2021-07-24:
In the ESP8266 Community Forum, someone informed us that: "Firebase switched to TLS 1.2 and AT firmware doesn't support it".

We still need to confirm it with Firebase, but it sounds reasonable.

mavishak, 2021-07-27:
We entered our database url in to: [https://www.cdn77.com/tls-test](https://www.cdn77.com/tls-test).
This is what we got
[image]